OpenSourceEBike/TSDZ2-Smart-EBike #1: Usage of the serial bootloader Motor and LCD3

shadow974, 2018-10-04:
The STM8 plattform has an embedded Bootloader which listens to commands in the very first seconds of the bootup-procedure. It has to be enabled by the option-bytes and is deactivated when read-protection is enabled (so no chance for first-time-usage)

https://github.com/gicking/stm8gal is a complete replacement for stm8flash and running on Linux and Windows (should do on mac, too)

Im going to test this on all of my available hardware (motor, LCD3, XH18).

The bootloader should be activated on any new installations...

endlesscadence, 2018-10-27:
First working examples:
https://github.com/endlesscadence/stm8-experiments

shadow974, 2018-10-30:
Why are you coding your own flash-tool? Is stm8gal not working on windows?

sbridger, 2018-12-06:
It would be potentially useful if the firmware has a serial commend that makes it jump into the bootloader.  I understand that the bootloader has entry points for this.

When using bluetooth powered by the bike, the bluetooth link (probably) is not up, before the natural power-on bootloader phase has ended.

geeksville, 2019-08-09:
btw: if this option bit was turned on for the TSDZ2, I think it would be not-hard to make a special build for the SW102 display that enables the following workflow (for wireless sw updates of the motor code):

* User uses the android/iOS DFU app to load "update-motor-firmware.zip" into the SW102 (we'd need to use a special build, because in order to have enough space for storing a full image of the new motor code).
* The appload turns off power to the bike (to ensure that we can reach the stm8 bootloader within its 1sec timeout) 
* The user presses the power switch on the SW102 to run the updater app build
* The appbuild quickly pings the motor with SYNCHR to keep the bootloader alive
* The appbuild installs the new motor firmware and marks the SW102 appbuild as invalid and renters the SW102 over the air BLE updating bootloader
* The user installs the new SW102 release which goes with the motor.

casainho, 2019-08-09:
The firmware on TSDZ2 is only about 25% free, so, no point to use bootloader.

See that original firmware is within the limit of 16kbybtes flash memory. We are using 32 kbytes even if the microcontroller reference says it is 16 kbytes.

geeksville, 2019-08-10:
Though I wasn't clear: the STM8 includes a bootloader in the mask rom which
is automatically started if the option bits are set to start it (so it
doesn't cost flash)

sbridger, 2019-08-11:
Enabling the bootloader in the display would also be (potentially more) useful, as the display doesn't have a connector for the programmer. Once bootloader was enabled the display can be reflashed through the serial connector. (and the display can be sealed up permanently again)

Probably also working for the STM32 based ones, as I understand they also use same type of bootloader.

As I said above, you could also add a serial command to jump into the bootloader

geeksville, 2019-08-11:
For the SW102 we've (where we is @lowperformer) have already enabled the
bootloader.  Only the initial programming requires opening the case.  All
further updates are done via Bluetooth from an Android or iOS phone.

The benefit of having the bootloader on for the motor is that after pushing
an update to the display, the display can then update the motor.

(Sent from a phone - please ignore typos)

casainho, 2019-08-28:
It is the motor firmware itself that controls the option bytes, so I think you should change on firmware only. I think that now on wiki there is no mention to the option bytes anymore.

geeksville, 2019-08-28:
Good point.  It can be set straight from the hex file.

(Sent from a phone - please ignore typos)