OpenSourceEBike/TSDZ2-Smart-EBike #3: Revert the assist levels to simple factors as it was previously

endlesscadence, 2018-10-05:
Quite a few users are asking whether we can revert to the old of having plain assistance factors instead of a power-like configuration.

jbalat, 2018-10-07:
I agree. The fact that you need to set 2000w to get 200w is really confusing. If we say level 1 has 1x multiplier then the user can easily work out whether he wants 2x 3x 4.5x etc on the other levels.
Keep in mind the levels are just a multiplier for the torque sensor reading so it just means that for 2x you only press half as hard.. If your adjusted torque value is >255 then it stays at 255.. At a 9x multiplier you would just be leaning your foot on the peddle to get it to 255.  The power overide function (on-ff/+ button) then sets the maximum power at 255 adjusted torque value.. Hope thats easy to follow

hurricane2, 2018-10-08:
Simple multiplier for sure. I explained it like this to friends like this and and they immediately got it. Superman legs on tap. Thanks for listening.

casainho, 2018-10-08:
I think is not a good idea, strategically, because:
- there is no mean for that factor, while it is now is for motor watts that means energy
- users want defines assist level as a factor of pedal human power in watts/energy

A good plan would be for now to keep as it is and next develop the code for reading pedals watts (I did that before and I can recover the code on g8t history) and plug the 2 values and so finally users would set the assist level as a factor of pedal power.

jbalat, 2018-10-09:
> I think is not a good idea, strategically, because:
>
> * there is no mean for that factor, while it is now is for motor watts that means energy
> * users want defines assist level as a factor of pedal [...]

If you look at my reply above then this still holds but the maximum torque (255) can be something the user can define. For instance sit on the seat and put pressure on the peddle which will define the most pressure you want to use. The will be used for 1x and 2x is half that... etc.

endlesscadence, 2018-11-11:
Fixed with #46

closed 2018-11-11 by endlesscadence