OpenSourceEBike/TSDZ2-Smart-EBike #6: Prevent drag when pushing the bike backwards

endlesscadence, 2018-10-05:
It seems that we're still powering the motor coils when cadence and speed is zero.

troebi2, 2018-10-09:
It seems to be released, when you give a little torque (just lay your leg softly on the pedals) on the pedals.

endlesscadence, 2018-10-10:
Thanks for the information! I'll try this myself as well.

sbridger, 2018-10-17:
A related non-software issue: when touring, and the system dies, what is the easiest way to field remove the blue gear to eliminate the drag?

casainho, 2018-10-19:
I found the reason for this issue!!

When we push the ebike backwards, the pedals rotate and also there the torque -- this is easy to very on LCD3 looking at pedal cadence and torque sensor values.
So, with cadence and torque sensor when we push the ebike backwards, motor will be powered as if we was doing force with our legs on the pedals.

The solution: I think we need to implement detection of pedals rotation and if the rotate backwards, we should set torque sensor value = 0 as also pedal cadence = 0.

endlesscadence, 2018-10-19:
@casainho Very nice, I'm able to fix and test this in the upcoming days.

casainho, 2018-10-19:
Grab the code from here, should be fast to have it done!!!

1. https://github.com/OpenSource-EBike-firmware/BMSBattery_S_controllers_firmware/blob/78bbbc4ca328e8bfe3d8414280a0583528237b3e/firmware/motor.c

2. https://github.com/OpenSource-EBike-firmware/BMSBattery_S_controllers_firmware/blob/78bbbc4ca328e8bfe3d8414280a0583528237b3e/firmware/ebike_app.c

endlesscadence, 2018-10-19:
Will do, thanks! 😄

endlesscadence, 2018-10-23:
@casainho Unfortunately this doesn't seem to work. I also don't see the torque increasing much when moving the bike backward. Sometimes the torque even decreases! There is hardly any change at all. I only see the motor speed (ERPS) increase very fast but I think this simply means that the motor magnets are moving? There is no change in drawn wattage too (on display).

casainho, 2018-10-23:
W9rks for me if I put a very high gear, that will make more torque and speed of the chainring...

endlesscadence, 2018-10-23:
Of course, I can imagine that it's possible to get some torque that way but all I am saying is that torque isn't the (only) issue if it's an issue at all...

casainho, 2018-10-23:
I get torque and cadence, then I see that PWM increases!! All this I see on LCD3.

Ranla1, 2018-11-12:
Hey, if the motor know that the pedals are turning backwards. Can you aktivet brakesensor?

casainho, 2019-04-07:
Done on 0.19.0.

closed 2019-04-07 by casainho