OpenSourceEBike/TSDZ2-Smart-EBike #7: Correct calculated value of Watts/hour

casainho, 2018-10-09:
_I tested the power consumption with a watt-meter this morning, here are the numbers :_

_LCD3 power usage : 60.6Wh
Turnigy watt-meter : 87Wh_

_Peak : 624W_

_So there is a difference. Hope it helps for future developments._

casainho, 2018-10-09:
Works well for SOC if: Simple full discharge your battery and take note on the counted Wh value on LCD3, then enter that value on LCD3 configuration and you will see that SOC works very well!!

I know that there is a time dependency on the calculation, as it is watts/hour. If the time is incorrect, then calcs will fail. And the time counting depends on the other code running for all the other tasks........

endlesscadence, 2018-11-11:
@casainho is this improved with https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/commit/7cf028c383e753942f4a732f728e4d213bf0dd08 ?

This issue can be closed?

casainho, 2018-11-11:
Yes. I will close.

closed 2018-11-11 by casainho