OpenSourceEBike/TSDZ2-Smart-EBike #11: Field weakening

casainho, 2018-10-12:
I think the idea is that we keep FOC working and get the best efficiency possible:

// 36V motor: L = 76uH
// 48V motor: L = 135uH
// ui32_l_x1048576 = 142; // 1048576 = 2^20 | 48V
// ui32_l_x1048576 = 80; // 1048576 = 2^20 | 36V

BUT when we approach the ERPS speed limit (and/or duty_cycle starts decreasing ???), we should adding a phase advance value so we start to loss efficiency but we get more motor speed, over the nominal max speed of 4000RPM/450ERPS.

I think by now you already have some good experience with this values... maybe we can figure out an algorithm to do this. And we can start by something very simple, like doing some iterations (increasing/decreasing of the angle offset until we get the motor_speed_erps we are looking for).

sbridger, 2018-11-26:
I measured 36V motor R accurately (4 point measurement at 1A)= 0.123ohm/2 = 0.061ohm/winding
(Previous based on incorrect assertion R~1/N , actually R~1/N^2)

I also checked L using single current slope 0 to 1A, getting 85uH. This however was not especially accurate, so 76uH value seems reasonable at Ioperating.

casainho, 2020-05-12:
Implemented on last firmware version 1.0.0.

closed 2020-05-12 by casainho