OpenSourceEBike/TSDZ2-Smart-EBike #41: km/miles counter

AndyME2, 2018-11-05:
What would be really useful is a counter, that accumulates distances until it is reset. Same as in a car or in the original firmware: you have a total ODO and you have a "trip" counter that does not start at zero everytime you start the bike, but it will add up until you reset it. This way you can easily control the distances that you are interested in. 
Reset must be easy, without menu. Is the arrow down/on-off button combo not free? I think this is a very useful feature that should be available.

casainho, 2018-11-05:
That is already implemented on the firmwware.

AndyME2, 2018-11-05:
sorry, i cannot see this.
you mean something else than submenu 8.3?
My trip meter always resets itself to 0 when the display goes off (0.14.2) 
and how would it be reset? by a key combination? to go into the menu each time is not really a good solution. maybe you can enlighten me here...

casainho, 2018-11-05:
There are 2 different distances shown on odometer field. One resets every power cycle and other only on menu configuration option xxx

AndyME2, 2018-11-05:
yes,  but this is absolutely not what i am talking about. sorry, that i have to disagree here with all due respect for you.

I am talking of a counter that 1. does not reset to zero at every power cycle, same as the total km counter, and can  2. be reset anytime EASILY by pushing a key combination e.g. down arrow and power, or two fast clicks etc etc. 
i think it is necessary to have a total counter that counts ALL km from start and in addition one that counts all km from last reset. exactly as in a car. and in a car you push a button and it is reset to zero, and in original firmware you also push a button and it is reset to 0.
I think this is an important feature in every day life.

casainho, 2018-11-06:
Ok, now I understand and you are correct -- there is no such counter. And yes, we need more buttons combination so we can do quick changes like reset total trip counter.

leon927, 2019-01-17:
This is solved in stable version 0.17.0 :)

closed 2019-01-17 by casainho