OpenSourceEBike/TSDZ2-Smart-EBike #49: Implement battery current ramp

casainho, 2018-11-09:
https://endless-sphere.com/forums/viewtopic.php?f=30&t=93818&start=1025#p1423861

I do not agree that we should just reduce the PWM_DUTY_CYCLE_RAMP_UP_INVERSE_STEP, as I believe it should be worst for the gears and other mechanical parts.

I think it can be done like this:
1. we should implement a current ramp and not a duty_cycle ramp. I think now as duty_cycle representing motor rotor speed and the current representing the torque.
2. implement: current ramp AND PWM_DUTY_CYCLE_RAMP_UP_INVERSE_STEP at a lower value maybe like 10
3. Let´s say we set our target current to 10 amps and the current ramp increases 1 amp per each 0.1 second, so after 1 second the motor ends at 10 amps but the current/torque stress on gears and mechanical parts increase with the ramp from 0 to 10 in 1 second.
4. At the first step on 0.1 second, the current ramp controller will set current to 1 amp and duty-cycle will jump very fast on that 0.1 second to get 1 amp and that should make motor rotate very fast to the desired final speed and engage the gears but using that low current... -- I think the result is what we all would like to have: fast response but not with an abrupt torque!!!

endlesscadence, 2018-11-13:
Implemented in version 0.16.0. This issue will be closed when the feature is tested and good feedback is received.

casainho, 2018-12-07:
Works very well on 0.16.

closed 2018-12-07 by casainho