OpenSourceEBike/TSDZ2-Smart-EBike #79: Menu navigation could be easier

j0bro, 2019-03-25:
Based on your excellent work I decided to go for this motor and I now have a super smooth ride to work. I am also a software engineer, and I noticed the button input is done other than I anticipated.

It would be nice if the up/down buttons of the KT-LCD3 button controller would perform up/down in the main and sub menus. The power button should act as the selection, always. It is currently confusing to me that I have to navigate up (only) in the menu using the power button and that pressing either up/down performs the selection.

Next to this I wondered if it is possible to print text in the odometer field using the 7-segment characters. How many characters are available? For the main and sub menus it would add a welcome level of feedback if (a portion of) the selected menu name is printed, like 'batt', 'dist', 'crui', 'offr', for example.

C is not my mother tongue, but I could try to see what changes are needed. Keep up the amazing work!

casainho, 2019-03-26:
I agree with menu navigation. Do you want to implement that improvement and submit a pull request??

 About showing text, forget about it because the project is closed since we are at memory limit...

You can jump to 850C color LCD or SW102 Bluetooth LCD:
- https://github.com/OpenSource-EBike-firmware/Color_LCD/wiki/SW102
- https://github.com/OpenSource-EBike-firmware/Color_LCD/wiki/Bafang-850C

Firmware already runs well on 850C color LCD.

j0bro, 2019-03-27:
Hey thanks yes I will see what I can do; and thanks for the heads up about those other displays. That SW102 looks pretty nice and compact, shame that it is hard to open! What is the Bluetooth normally used for in that one?

casainho, 2019-03-27:
Bluetooth is used to communicate with mobile app. On that app we could configure the system as we do on LCD3.

We could also communicate with the battery BMS and read each cell voltage, etc, here: https://opensourceebikefirmware.bitbucket.io/development/Smart_BMS_with_bluetooth.html

j0bro, 2019-03-27:
Aha I see; such a mobile app sounds quite useful. Is it just an idea or already in development? I have extensive experience in mobile app development and connectivity (including BLE).

casainho, 2019-03-27:
No one is developing a mobile app nor the firmware for Bluetooth.

I added the first code that builds for this NRF51 and this LCD, see here: https://github.com/OpenSource-EBike-firmware/SW102_LCD_Bluetooth

My plan is to try flash and debug that code.

j0bro, 2019-04-16:
Hey, I was trying to setup my dev environment; I am working on a Mac. sdcc was easy to install but when compiling, I seem to need the stm8-size binary, among others. Can you tell me to what package that binary belongs (it is not part of the sdcc package).

casainho, 2019-04-16:
I think you are missing binutils. I do follow this guide and I hope it works for you: https://stm8-binutils-gdb.sourceforge.io/

j0bro, 2019-04-16:
Right, that worked, thanks. Was able to patch and build binutils-2.30 from source using the supplied script; although now the actual compilation fails with a SIGABRT on src/display/KT_LCD3/lcd.c

Do you use a specific version of sdcc?

This is the output and the command just before the error:
```
sdcc -V -c -I../../common/STM8S_StdPeriph_Lib/inc -I. -I../../ -mstm8 -Ddouble=float --std-c99 --nolospre --out-fmt-ihx --debug  -olcd.c lcd.c
+ /Users/<snip>/bin/sdcc/bin/sdcpp -nostdinc -Wall -std=c99 -I../../common/STM8S_StdPeriph_Lib/inc -I. -I../../ -Ddouble=float -obj-ext=.rel -D__SDCC_STACK_AUTO -D__SDCC_CHAR_UNSIGNED -D__SDCC_MODEL_MEDIUM -D__SDCC_INT_LONG_REENT -D__SDCC_FLOAT_REENT -D__SDCC=3_9_0 -D__SDCC_VERSION_MAJOR=3 -D__SDCC_VERSION_MINOR=9 -D__SDCC_VERSION_PATCH=0 -D__SDCC_REVISION=11195 -D__SDCC_stm8 -D__STDC_NO_COMPLEX__=1 -D__STDC_NO_THREADS__=1 -D__STDC_NO_ATOMICS__=1 -D__STDC_NO_VLA__=1 -D__STDC_ISO_10646__=201409L -D__STDC_UTF_16__=1 -D__STDC_UTF_32__=1 -isystem /Users/<snip>/bin/sdcc/bin/../share/sdcc/include/stm8 -isystem /usr/local/share/sdcc/include/stm8 -isystem /Users/<snip>/bin/sdcc/bin/../share/sdcc/include -isystem /usr/local/share/sdcc/include  lcd.c 
Caught signal 6: SIGABRT
make: *** [lcd.rel] Error 1
```

casainho, 2019-04-16:
SDCC has some issues on STM8. I suggest you to try 3.9.0 that was released yesterday. If it works for you, I will also switch to it.

I use SDCC 3.8.0 SVN revision #10557.

j0bro, 2019-04-18:
I now tried setting up the whole toolchain on a Ubuntu VM too (indeed using SDCC 3.9.0 final), and that fails on compilation of lcd.c as well (SIGABRT, signal 6). I even tried updating the STM8S_StdPeriph_Lib to 2.3.1, but to no avail. Is there any way to get more verbose output from the compiler as to why it crashes?

My last resort is now to check with a Windows VM, but I don't like that it doesn't work on Mac or Linux because Windows is just a PITA when it comes to using POSIX tools (Cygwin etc.)

casainho, 2019-04-18:
Then try SDCC 3.8.0 SVN revision #10557, that is the only version I tested that worked for me. The final stable 3.8.0 version does not work for me but the SVN revision #10557 does.

j0bro, 2019-04-25:
Took the installer from here ([SDCC 3.8.0 rc](https://sourceforge.net/projects/sdcc/files/sdcc-win64/3.8.0/sdcc-3.8.0-rc1-x64-setup.exe/download)1) and only that one managed to compile the sources successfully. The mac version of 3.8.0-rc1 does not even work. Writing this from my Windows 10 VM 😄

Finally I am able to start coding, will keep you posted.

casainho, 2019-04-25:
Well, right now we have SW102 working and we are 3 developers for it!! I suggest you to forget LCD3 and focus on SW102 -- the only disadvantage is that SW102 is smaller screen but is way more flexible on the information we can show!!!

j0bro, 2019-04-25:
Hah that's an unfortunate timing :) I only recently installed my TSDZ2 motor + KT-LCD3 and so far it is very good. The SW102 display looks nice though; but must it be opened to be able to flash it? The code looks very pre-alpha. Good potential! But until there is some usable functionality for it, I am happy to keep on using the LCD3 and tinker with that a bit. I am sure most users actually have it right now.

j0bro, 2019-04-25:
@casainho could you provide with me write permissions to the repo? I can't push my feature branch to origin.

> 
> > git push -u origin feature/79_menu_navigation
> remote: Permission to OpenSource-EBike-firmware/TSDZ2-Smart-EBike.git denied to j0bro.
> fatal: unable to access 'https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike.git/': The requested URL returned error: 403

casainho, 2019-04-26:
I just merged your pull request! Thanks!!

casainho, 2019-04-26:
Can we close this issue now?

j0bro, 2019-04-26:
Of course, thanks!

closed 2019-04-26 by j0bro

ashleysommer, 2019-05-02:
@j0bro @casainho 
I too am a software developer and recently installed my TSDZ2+KT-LCD3. I've got my dev environment set up on my Linux PC and I've produced a couple of custom builds. I have no intention of moving to a different LCD any time soon, so I'm keen to keep this project up and running on KT-LCD3 with just bugfixes and performance improvements, of course no new features.