OpenSourceEBike/TSDZ2-Smart-EBike #81: Battery profiles

casainho, 2019-04-07:
y la8rat » Apr 06 2019 1:41am

casainho wrote: ↑Apr 06 2019 12:22am
la8rat wrote: ↑Apr 05 2019 7:37pm
i know we are out of space for new code on the ktlcd3, but on future displays it would be great to have a battery profile menu.
where everything needed to swap to a different voltage, capacity of battery was able to be saved. and switched between with ease.
So you need that? Can you please explain why and how are doing it??
good morning

i have 3 batteries, 13s3p 13s5p and a 14s5p i swap between the batteries often for short fast ride i use 13s3p the others for longer rides. they all obviously need different setting that are spread across a few different menus. 
i find i forget the various setting and what menu they are in.
how to implement? not sure maybe a nested sub menu for battery profiles names then lower menu for settings per battery.

wattzor, 2019-09-15:
quick side question, If i get the 48v 750w motor. Can i make the firmware of it run on 36volts ? 
i guess i loose some preformens due to motor lindings but, will this work for a while ?

Frans-Willem, 2019-09-16:
I've got a 36V and a 48V motor at home, having tried both with a 52V battery, I highly prefer the 36V motor. If you intend to run at both 36V and 48V, I'd recommend getting the 36V motor. the 36V will work fine on 48V, albeit a teensy bit less efficient, while the 48V motor will likely be very very very sluggish at 36V.

wattzor, 2019-09-16:
Oh okey thanks! I already order the 48v one but i guess i get a bigger battery then! Or will the controller handle 72 volt? (Two 32v in serie)