OpenSourceEBike/TSDZ2-Smart-EBike #87: Throttle not responsive

casainho, 2019-04-22:
Tested in v0.18.2 and throttle input is used and works as expected.
Configured using menu item 8 sub menu 0 set to value 2

Tested in v0.19.3 beta and verified throttle ADC readings are fine under menu 10 sub menu 0, But no motor movement.
Enable throttle menu has changed to menu item 8 sub menu 4 set to value 2 confirmed this is correct.

Pscope, 2019-04-25:
This appears to still be an issue in v0.19.5 beta

leon927, 2019-06-11:
I have maybe missed this but is it solved in the 0.19.0 beta 8? I have no throttle to test with. Have not heard anyone having issues.

Pscope, 2019-06-12:
> <TSDZ2-Smart-EBike@noreply.github.com>
> responsive (#87)

Hi Leon,

Sorry I am unable to test this at the moment, as the trike with the throttle 
is now with the owner, not at my location, but hopefully in the next few 
days i will have access to it to try . I will test and let you and the 
thread know. 

Cheers

Neil

leon927, 2019-06-12:
Neil, thank you!

Cheers 

Leon

leon927, 2019-06-18:
Neil, 

Sorry to bother you but have you had the chance to test the throttle? If not, it is totally fine. Just checking.

Leon

Pscope, 2019-06-18:
Hi Leon,

Sorry but no not yet :(

Unfortunately the trike is currently in a locked store I dont personally 
have the key for but i know the man who does!

I hopefully will get access to it this weekend, and will bring it home. I 
haven't forgotten, Its just not in my possession at the moment 
unfortunately. I was going to add a throttle to my own TSDZ" but I am 
running the temperature sensor so I have to open the motor to disconnect 
that to test a throttle, and well I don't really want to do that if I can 
help it.

Feel free to keep asking, I have a lot of projects on at the moment so i'm 
happy to receive a gentle reminder :)

all the best

Neil

PS are you UK based?

Date: Tue, 18 Jun 2019 06:14:10 -0700
responsive (#87)

leon927, 2019-06-19:
No problem at all and absolutely no need to install the throttle! Will also ask the thread to see if anyone has tried it out.

Based in Sweden, slightly north of you guys! ;)

Leon

Pscope, 2019-06-20:
i have the trike at home now so will test tonight and get back to you. 

cheers

Neil

  _____  

Cc : Pscope [github@perryscope.com], Mention
[mention@noreply.github.com]
Date : Wednesday, June 19 2019 14:03:30

<https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/issues/8
7?email_source=notifications&email_token=AJIX4SKUHM62WCQ3K5P6YM3P3IVCFA5
CNFSM4HHNA6WKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZG
GLP3IVCFANCNFSM4HHNA6WA> .
<https://github.com/notifications/beacon/AJIX4SKWN47F4M4FORKAKBDP3IVCFA5
CNFSM4HHNA6WKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZG
ODYBZKMY.gif>

Pscope, 2019-06-20:
Hi Leon,

OK I have managed to test the throttle on 0.19.0 beta 7 and I'm afraid its 
the same.

If I now go to menu 9 item 1 I can see the throttle value change from 0 to 
255 as i twist it, so its being detected by the firmware. but nothing 
happens.

downgrade to the last release 0.18.2 and it works.

so something in 0.19.0 has stopped it.

its been the same through each beta release.

Cheers

Neil

Date: Wed, 19 Jun 2019 06:03:30 -0700
responsive (#87)

leon927, 2019-06-21:
Appreciate you took the time to test!

Did you mean beta 8? I think it does not matter if it was beta 7 as the problem should be present in beta 8 as well. But just want to make sure.

I will take a look at this and once again, thank you for testing!

Leon

Pscope, 2019-06-21:
Hi Leon,  I did actually checked beta 7 last night not beta 8. I missed that 
release! doh. will check that also tonight, but like you say I don't think 
there are any changes 

Cheers
Neil

Date: Fri, 21 Jun 2019 05:09:29 -0700
responsive (#87)

Pscope, 2019-06-21:
sucess!

0.19.0 beta 8 thtottle now works!

i will do some more testing but yes its looking like its fixed.

cheers
Neil

  _____  

<https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/issues/8
<https://github.com/notifications/beacon/AJIX4SKWN47F4M4FORKAKBDP3IVCFA5

leon927, 2019-06-21:
That is actually great news! Please do test some more and let me know if you are satisfied!

And if you could make sure and double check that it is the 0.19.0 beta 8 that is uploaded both in the display and controller that would be much appreciated. 

Can not thank you enough for the feedback and testing, Neil!

Cheers,

Leon

leon927, 2019-06-24:
@Pscope - Big thanks for the help Neil! The issue is solved!

@casainho - We can now close this as it is solved in the latest beta and thereby the stable 0.19.0.

closed 2019-06-24 by casainho