OpenSourceEBike/TSDZ2-Smart-EBike #93: Error code 1 happens frequently when it should not

casainho, 2019-05-09:


arnoutzw, 2019-05-11:
Hi Casainho, I noticed this error popping up also a lot, do you have already a clue what might be causing it?

Thanks!

Btw, I really like the opensource firmware and maybe hope to contribute in the future!

Best regards, 

Arnout

Dirkkro, 2019-05-18:
This is  is the code when you have force on the pedal, and the pedals are not moving.  after  some seconsd the  power is swiched off. 
As a workaround you can  try to diable  "power without pedal rotation"  You may be supprised  how fast the engine is pushung in a  very small fraction of rotation !

leon927, 2019-06-24:
@casainho - We can now close this as it is solved in the latest beta and thereby the stable 0.19.0.

closed 2019-06-24 by casainho