OpenSourceEBike/TSDZ2-Smart-EBike #94: Motor assistance does not resume after standstill when lights are on

j0bro, 2019-05-15:
I installed lights and now when I have them on, the motor does not resume assistance after you had a standstill. It looks like you need to keep pedal cadence going, or some speed or power output on the motor, I don't know what makes it stop exactly.

After I switch off the lights, motor assistance kicks in again when pedaling.

j0bro, 2019-06-11:
Is there any workaround possible for this? Thanks in advance @casainho and @leon927

leon927, 2019-06-11:
The best thing to do is to just wait for the stable 0.19.0 that is soon to be released. And if you still experience problems in that version it will be sorted out. 

If you can not wait for the official release I can recommend the 0.19.0 beta 8. No known bugs and has a lot of great changes.

j0bro, 2019-06-11:
I found this issue has been (re)visited already a few times in, but for me, the fix as proposed in closed issue #71 does not work. Should we maybe have a configurable lights current to allow users to tweak this behaviour?

leon927, 2019-06-11:
It is best to first update to the latest version and see if the problem remains. And if it does we will try to solve it. I think the stable version 0.19.0 will be released in a couple of days :)

closed 2019-06-24 by casainho

j0bro, 2019-06-25:
Updated to 0.19, seems it works now, thanks!