OpenSourceEBike/TSDZ2-Smart-EBike #101: Battery SOC percentage wrongly displayed after power cycle

j0bro, 2019-06-11:
I configured the KT-LCD3 to display the SOC as percentage in the lower right corner of the ODO field. After a full charge this displays 100, and during the ride this value decreases accordingly. 
At power up for a next ride the value mostly starts at a completely wrong value, around 40 which makes we wonder if it is not displaying the voltage as raw value instead of converted to charge percentage.
This is totally reproducible.

leon927, 2019-06-11:
What version of the software are you using?

j0bro, 2019-06-11:
Ah sorry forgot to mention; I am on the 0.18.2 firmware.

I rode home today from work and the same thing again - it resumed decreasing not from the actual percentage but somewhere around 40-ish. It went down to zero and stayed there. The battery indicator does show the right SOC.

casainho, 2019-06-11:
Please read carefully the wiki to make sure you did the correct configuration about that SOC field.

leon927, 2019-06-11:
Will try to reproduce the problem. 

Will charge above the voltage threshold so everything is reset. Will ride the bike until I get below 100 percent. Then power off and see what happens.

Will test on the 0.19.0 beta 8. But the calculation is the same compared to 0.18.2.

j0bro, 2019-06-11:
I set menu 2:0 to value 1 (display watt-hours consumed and percentage of charge remaining ) and menu 6:0 to value 2 (Battery state of charge in percent). I have a 36V battery and just double checked while it is on the charger - I set the threshold to 41.5V but even when the display showed 41.5V or even 41.6V, the value for the SOC percentage was not resetting to 100.

I expected the value to rise while charging; like the battery indicator does even. But for some reason it is behaving erratically. Thanks in advance for testing!

j0bro, 2019-06-11:
[image]
This is what my display shows while charging. I set the battery-full-threshold to 41.5V and the right bottom odometer field to show the SOC percentage. See that the battery indicator (left top) does show correct value.

leon927, 2019-06-11:
No problem! :)

The SOC is only reset when you power up the display. It is a one-time check done during startup. This is why it did not reset when you have it charging and looking at the value. 

Charge the bike, power on and then it is reset to zero.

j0bro, 2019-06-11:
I did power cycle the display but to no avail. The value indeed remains zero, but I expected it to be set to 100 (charge went above the threshold). It would be a good thing IMHO if that SOC field is updated continuously, just like the battery indicator on the left-top. I could see if I can make a PR for that if desired.

leon927, 2019-06-11:
Just tested it and it works as expected on my short test. Could not reproduce it on version 0.19.0 beta 8, very strange.

What you can do is await the official release of the 0.19.0 version. But everything should work with the 0.18.2 also as no one has reported "no-reset".

Try once again by setting the voltage threshold to 36 volt. Power down the bike and then power back on. See if it resets. Do not have it plugged in and charging when you test this.

j0bro, 2019-06-11:
Ok I lowered the threshold and had the charger unplugged, power cycled and saw the value go back to 100 again. But that was not the issue - it should resume showing the correct SOC percentage after several bike rides and that is where it goes wrong for me. Will keep you updated!

leon927, 2019-06-11:
I understand that was not your original issue but that one is easy to test right now. Try it out some more and let us know how everything works.

leon927, 2019-06-14:
Hi @j0bro 

Have you tested it some more and does it work as designed?

j0bro, 2019-06-14:
Hey Leon,

Yes I rode the bike twice now and even made pictures, but after setting the
'full' threshold to something a bit lower than what I had, the issue seems
to be resolved. I think that still a live update of this value would be a
good idea, but I can live with it as it is. Please go ahead and close!

Thanks, Jeroen

leon927, 2019-06-24:
@j0bro - Thanks for the comments and feedback Jeroen. By the way, glad to hear it is solved!

@casainho - We can now close this.

closed 2019-06-24 by casainho

leon927, 2019-07-07:
> Hey Leon, Yes I rode the bike twice now and even made pictures, but after setting the 'full' threshold to something a bit lower than what I had, the issue seems to be resolved. I think that still a [...]

@j0bro The SOC is now reset live once it detects battery voltage over the threshold voltage. Should be closer to what you wished for. Look for it in the next version!

Leon