OpenSourceEBike/TSDZ2-Smart-EBike #103: ooh cool - need any help?

geeksville, 2019-06-29:
Hi,

I just ordered a TSDZ2 and LCD3 (though a LCD5 is apparently included in my bundle from China) for my bike and I'm going to give your cool project a shot!  I'm a reasonably experienced embedded sw/hw developer - do you have any easy issues that need work? (to give me a goal in my initial experimenting).  Is this project still in active development?

casainho, 2019-06-29:
I think the next important thing is to make the new LCD working, I think will be the favorite of the users:

https://github.com/OpenSource-EBike-firmware/Color_LCD/wiki/Bafang-LCD-SW102

closed 2019-06-29 by casainho

geeksville, 2019-06-29:
ok sounds good.  I'll order a SW102 to poke around with.  Is this the right branch? https://github.com/OpenSource-EBike-firmware/SW102_LCD_Bluetooth/commits/SDK_12.3.0 and @lowPerformer is the person mostly trying to make this thing go? ;-)

casainho, 2019-06-29:
Yes and yes. I did agree to port the configurations menu from 850C LCD that I did to SW102 LCD bit I am not finding free time. Maybe you could look at it?? The configurations menu is the next logic step

geeksville, 2019-06-30:
sure thing, though I probably won't start work on this project for about 3 wks (which is what my various packages from China (including the SW102 I just ordered) will take to arrive).