OpenSourceEBike/TSDZ2-Smart-EBike #104: bus error while compiling LCD3/lcd.c with sdcc 3.9.0

geeksville, 2019-07-04:
Hi,
The other C files in this project build just fine, but sdcc version 3.9.0 is crashing while compiling lcd.c on my x86_64 linux system.  What version of sdcc are other devs using?  The wiki just says 3.7.0 or newer.

```
kevinh@kevin-server:~/development/ebike/TSDZ2-Smart-EBike/src/display/KT-LCD3$ make -f Makefile_linux 
sdcc -c -I../../common/STM8S_StdPeriph_Lib/inc -I. -I../../ -mstm8 -Ddouble=float --std-c99 --nolospre  --out-fmt-elf --debug  -olcd.c lcd.c
Makefile_linux:73: recipe for target 'lcd.rel' failed
make: *** [lcd.rel] Bus error (core dumped)
kevinh@kevin-server:~/development/ebike/TSDZ2-Smart-EBike/src/display/KT-LCD3$ sdcc -v
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ez80_z80/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8/pdk14/pdk15 3.9.0 #11195 (Linux)
published under GNU General Public License (GPL)
```

Frans-Willem, 2019-07-04:
From my experience, SDCC isn't all that backwards or forwards compatible, and not as stable as for example GCC.
If the wiki says 3.7.0 or higher, I'd wager it was tested on 3.7.0, did you try that one ?

geeksville, 2019-07-04:
Thanks.  Yep, I just tried 3.8.0 and it is fine.  I'll send in a PR with updates to the wiki saying 3.9.0 is dame.

closed 2019-07-04 by geeksville