OpenSourceEBike/TSDZ2-Smart-EBike #107: [Request] Pedal-torque threshold for 0 cadence starting.

Frans-Willem, 2019-07-05:
Coming from the stock firmware, I feel like the starting from a traffic light is much much slower. I get that everything else feels more agile, but sometimes I feel like it takes almost a full rotation of the pedals before I feel the power kicking in.

Things I've tried so far:
- Enabled startup boost. Does give a boost, but still slow to kick in.
- Current ramp to 10amp/sec. Again, once it starts, it starts faster, but it still takes relatively long to actually start.

The only thing that really works is "Motor assistance startup without pedal rotation", with the downside that whenever I rest my foot very slightly on the pedals, it already starts twitching.

Would it be possible to replace "Motor assistance startup without pedal rotation" with a "Motor assistance startup torque threshold without pedal rotation", which would result in logic like:
- If the pedals are rotating, always assist.
- If the pedals are not rotation, assist whenever the pedal torque goes over a certain threshold.

That way if you set a reasonable threshold, you can still rest your foot on the pedal without starting the motor, but once you actually put some weight into it, it should start right away instead of waiting for rotation.

Frans-Willem, 2019-07-05:
More info:
I've also confirmed that both my torque sensor and my cadence sensor are working through the "Advanced Technical Data" menu.

leon927, 2019-07-05:
Exactly the issues you mention is something I am working on. Please be patient and I will do my best to give a system response and feeling equal to or much better than the original firmware! Have made some great progress today actually and it will only get better! Would appreciate your feedback whenever I am finished!

Frans-Willem, 2019-07-06:
After having gone through the (19.0) code, I have a suspicion on what's happening.
This firmware checks cadence RPM, which is calculated from the time between your pedals passing the PAS magnets. If you've been standing still, the time from your last magnet hit to your first magnet hit will be so large, that the RPM will still be 0. It's only after you'll hit your second magnet, that the RPM reflects your cadence.

I think that stock firmware may not actually be checking the cadence, but just keeps track of how long ago your pedals passed a magnet.

Thus the open-source firmware, you need to pedal enough to pass two magnets before it'll allow the motor to start, whereas on the stock firmware, it's enough to pass just one magnet.

leon927, 2019-07-06:
The calculation of human power requires the actual angular velocity and not just measuring how long ago a "hit" was. So what you are proposing makes a drastic difference on the firmware. Not saying it is not correct just stating the fact.

In the not so distant future there will be a beta where you can test the changes I have made, and already planning to make, for your and others feedback on this issue.

Demion, 2019-07-06:
I also noticed problem with slower start and pedal twitching.

As I understand this is exactly implemented in **apply_torque_assist**. 
I know its WIP, but only thing I think should be implemented is transition from **TORQUE_ASSIST_MODE** to **POWER_ASSIST_MODE** when **ui8_pedal_cadence_RPM > threshold**. 
Maybe I missed that part.
https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/blob/9d5eaa788c44c4b9dccbd8458e13a9d03289c0f7/src/controller/ebike_app.c#L282-L306

leon927, 2019-07-06:
Slower start and twitching will be hopefully solved for all riding modes. Am working hard to improve this among other things.

geeksville, 2019-08-09:
I too have noticed this on my bike (which is a recumbent, so therefore on steep hills quick response of start-assist would help a lot).  @leon927 - If you need an alpha test user for your change I'm eager and willing.

leon927, 2019-08-10:
Yes, I would like some feedback and if you are willing that would be great!

Here is a link to the Alpha versions I have released:
https://drive.google.com/open?id=1CPGT-V1ZX7FCzfhgbrPMYfbYtY4_JH2R

Will release an Alpha 4 today hopefully so wait for that one if you want the latest news ;)

Here is the wiki:
https://github.com/OpenSource-EBike-firmware/TSDZ2_wiki/wiki/0.20.0-(DEVELOPMENT)-%7C-KT-LCD3-%7C-TSDZ2-%7C-Manual

geeksville, 2019-08-10:
@leon927 great!  I'll try it this week.  is the comms compatible with 0.19 displays?  I'd like to use my sw102 build - which happily talks to my 0.19 motor.

leon927, 2019-08-10:
I look forward to your feedback! Am finishing up on the Alpha 4 so will be waiting for you :)

Sadly no, it will take some work to get everything in sync. We have several riding modes now and a lot of new things in the 0.20.0. Tried to develop it as quickly as possible to have it ready for the new displays. Seems we are getting there at the same time.

Here is the 0.20.0 branch:
https://github.com/leon927/TSDZ2-Smart-EBike/tree/testing-pwm-acc

geeksville, 2019-08-13:
ok @leon927 I think I might wait a little bit because I've already got a queue of SW102ish/850Cish things that look fun to play with.  Then later we can update that codebase to speak the new 0.20 magic packets ;-)

casainho, 2019-08-13:
I think we should skip V0.20.0 because there is already a feature for V0.21.0 that needs space on display that maybe will not fit in KT-LCD3 -- the torque sensor full calibration. Also because I think we need to improve more the communications, with a specific packages for configurations to be sent only once at boot and after change configurations. All of this, I am afraid KT-LCD3 may not have enough memory but the new LCDs have.

Full torque sensor calibration is already developed and I am using it on my bicycles, so should be fast to add to V0.21.0. And for me, this new version could have only this new feature as also bugs corrections, to be quick to release this new version.