OpenSourceEBike/TSDZ2-Smart-EBike #108: Question about uninstalling firmware

hydriniumh2, 2019-07-09:
Hi, 
First off, wanted to say this project is amazing and I'm looking forward to installing it. 
However, I was wondering about if the installation process is reversible, for example if I wanted to go back to using the stock firmware to sell a used tsdz2. If so, are there instructions anywhere on how to do it / stock firmware to download?
Apologies if this isn't the right place to ask this.
Thanks!

Frans-Willem, 2019-07-10:
Eco Cycles have a repository of the stock firmware for all different versions here:
https://www.eco-ebike.com/blogs/eco-cycles-instructionals/tsdz2programmingfromscratch

You can use it to reprogram your TSDZ2 to a different type/wattage, or just to go back to stock :)

hydriniumh2, 2019-07-10:
Awesome, thats exactly what I was looking for. Thanks!

closed 2019-07-10 by hydriniumh2