OpenSourceEBike/TSDZ2-Smart-EBike #110: Official support from a Chinese manufacturer

geeksville, 2019-08-09:
So as I've been futzing on this fun project I find myself thinking about how things worked on another project I worked on (ardupilot).  Our software stack ended up getting used by various Chinese mfg as their standard preinstalled software.  Which was fine with us because it made a bigger base of users and a bigger collection of devs contributing fixes.

Have you considered talking with whoever makes the TSDZ2 to see if they would be interested in this?  I bet they pretty easily could make a new SKU with:

* TSDZ2
* SW-102
* correct software preinstalled in both devices (with bootloader to allow future sw updates for both the motor and the display over the bluetooth DFU in the SW102)

This would save them SW eng cost, give a better product with all  of ya'lls enhancements.  Users would know they would always have sw updates available.  Of course the mfg would need to do a deal with whoever makes the SW-102 but they would end up with a nicer product than their current dated LCD5 product line...

I bet a product manager in Shenzhen might find this quite appealing.

casainho, 2019-08-09:
The firmware on TSDZ2 is only about 25% free, so, no point to use bootloader.

See that original firmware is within the limit of 16kbybtes flash memory. We are using 32 kbytes even if the microcontroller reference says it is 16 kbytes.

I guess the manufacturer would not risk to use our firmware.

Anyway, I already sent some e-mails to TongSheng and I got no answer at all. Also, we have that USA companies that are selling TSDZ2 and KT-LCD3 with our firmware and they are in talks with the various manufacturers. This companies are asking since always for 850C and SW102 displays :-)

geeksville, 2019-08-10:
sounds good.  Though it might be worth turning on the bootloader anyways - because it is included 'for free' in the mask ROM of the STM8 (i.e. doesn't cost any flash).  If we want to have the option of someday updating the motor with the help of the display controller.

casainho, 2019-08-10:
Ah, okok, that would be good!!

leon927, 2019-08-10:
I considered asking you, @Casainho, if you have been in contact with them for future updates, changes and what not. But I guess you just answered that question: no response what so ever from them...

OT: You worked on Ardupilot, @Geeksville? That is a great project and I absolutely love that firmware! Have not looked into the development and it is a long time since I used it but always thought the developers did an amazing job. So I guess some of the acknowledgment should go to you!