OpenSourceEBike/TSDZ2-Smart-EBike #111: If motor stops receiving packets from display, we should do a few things for safety:

geeksville, 2019-08-13:
Per discussion with @casainho and @lowPerformer.  Capturing here to do someday:

If the motor stops receiving packets from the display, we should turn off a couple of modes the display might have forced:

* Turn off walk assist
* Turn off cruise mode

Because otherwise a sw (or hw, but more likely sw ;-) ) fault in the display controller could cause hangage (even once we enable the watchdog if we have user visible fault screens enabled).  We should be very conservative before taking this drastic step though - i.e. only if we miss 5 seconds worth of packets or something.

It is an open question if the pedal boost behavior should stay on?

casainho, 2019-08-13:
This is really important!! I am afraid of walk assist or cruise mode on that situation!!

leon927, 2019-08-15:
I have actually implemented that in the 0.20.0. If a packet is not received it will turn off everything.

The controller needs constant communication and if this does not happen it will go to OFF_MODE.

Here are the modes:

#define OFF_MODE 0
#define POWER_ASSIST_MODE 1
#define TORQUE_ASSIST_MODE 2
#define CADENCE_ASSIST_MODE 3
#define eMTB_ASSIST_MODE 4
#define WALK_ASSIST_MODE 5
#define CRUISE_MODE 6
#define CADENCE_SENSOR_CALIBRATION_MODE 7

But the throttle does work regardless. And maybe this would be good to disable as well?

shadow974, 2019-09-02:
I agree with "walk assist" and "cruise mode" as they are initiated by serial communication and cannot be stopped without it.. Throttle and last active assist mode should be remain enabled imho.