OpenSourceEBike/TSDZ2-Smart-EBike #112: 0.19 performance worse than 0.18

j0bro, 2019-08-26:
I am very hesitant to jump to conclusions and therefore am only describing a perceived correlation, but since the update to 0.19 I am experiencing some nasty issues with my bike:
- battery drains much faster
- motor makes a rattling noise (as if the gear is stripped or a bearing cracked but I already opened the drive assembly and all is fine!). When motor assist is off the noise is not there.
- the maximum power it reaches is only about 360W while the limit is set to 500W. It definitely also feels weaker at a standing start and when applying force to accelerate.
- I have had an occasion where the motor voltage type was set to 0 (menu 7:0:0, 48V) while I have a 36V motor. I did not set this myself. This made another strange which did go away by setting the correct voltage type again.

Is any of these issue known or experienced by others?

casainho, 2019-08-26:
I don't use 36V motor on my ebikes so I can't comment.

leon927, 2019-08-26:
@j0bro Regarding the efficiency: if I remember correctly there were some improvements in the calculation of energy consumed so it will seem as it is draining the battery faster but actually there should be no difference in efficiency. 

All other things you mention can be some mechanical problem that is not related to software. Or maybe related to having wrong motor type selected.

If you want to try out the 0.20.0 Alpha 8 I would love to hear some feedback from you before there is a stable release. 0.20.0 Alpha 8 is verified as much as 0.19.0 was and should be extremely stable with many improvements you might like. Soon there will be an Alpha 9 which is maybe the last version before the first official beta is released on the official project page.

j0bro, 2019-08-27:
Hey thanks for your replies guys, much appreciated. I also have some insights back for you.
Since I did think of a mechanical issue too at first I got in doubt when I saw that without motor assist, there was no apparent issue. I went on thinking it could be some on-off behaviour in the assist algorithm, making it 'feel' as if a gear is blocking.

In the end I went through the motor controller settings and saw the internal resistance value set to 130 (mOhm). I already suspected my battery (no name 18650 cells) to have degraded a bit, so the chemistry changed and with that the internal resistance. I changed the value to 180 (just a random choice) and already, the behaviour of the bike completely changed for the better. There is no apparent voltage sag, range estimation is stable, no more rattling noise, etc.

I will try to obtain the actual internal resistance value on my workbench and see if I get even better results with that.

leon927, 2019-11-18:
This can be closed.

closed 2019-11-18 by casainho