OpenSourceEBike/TSDZ2-Smart-EBike #113: Control motor current instead of battery current

AckmaniacSoftware, 2019-09-03:
Wouldn't it be better to have control about the motor current instead of the battery current. Because then you have direct control about the torque the motor generates. i see that the max motor current is set to 30 so it could be a smooth transition to watts or battery current once the max speed of the motor is reached.
Sadly i don't have the motor yet but when i see the code it seems that the feeling later on will be very powerful and a bit jerky at low speeds and less powerful at higher speeds. Even if you apply the same torque all the time.
I know i sound stupid because i did not ride the motor yet but i come from electric longboards where i optimized the firmware for hundreds of hours.
there i introduced the watt control mode which provides exactly the described behavior.

casainho, 2019-09-03:
Well, motor current is left constant while someone want to start play with it. Maybe you can be the developer to try improve the behavior, make tests and then propose your change.

AckmaniacSoftware, 2019-09-04:
here is my example of the apply_torque_assist methode,

`static void ebike_control_motor (void)
{
  // reset control variables (safety)
  ui16_duty_cycle_ramp_up_inverse_step = PWM_DUTY_CYCLE_RAMP_UP_INVERSE_STEP_DEFAULT;
  ui16_duty_cycle_ramp_down_inverse_step = PWM_DUTY_CYCLE_RAMP_DOWN_INVERSE_STEP_DEFAULT;
  ui8_adc_battery_current_target = 0;
  **ui8_adc_motor_current_target = ADC_10_BIT_MOTOR_PHASE_CURRENT_MAX;**
  ui8_duty_cycle_target = 0;

  // reset initialization of Cruise PID controller
  if (ui8_riding_mode != CRUISE_MODE) { ui8_cruise_PID_initialize = 1; }
  
  // select riding mode
  switch (ui8_riding_mode)
  {
    case POWER_ASSIST_MODE: apply_power_assist(); break;
    
    case TORQUE_ASSIST_MODE: apply_torque_assist(); break;
    
    case CADENCE_ASSIST_MODE: apply_cadence_assist(); break;
    
    case eMTB_ASSIST_MODE: apply_emtb_assist(); break;
    
    case WALK_ASSIST_MODE: apply_walk_assist(); break;
    
    case CRUISE_MODE: apply_cruise(); break;

    case CADENCE_SENSOR_CALIBRATION_MODE: apply_cadence_sensor_calibration(); break;
  }
  
  // select optional ADC function
  switch (m_configuration_variables.ui8_optional_ADC_function)
  {
    case THROTTLE_CONTROL: apply_throttle(); break;
    
    case TEMPERATURE_CONTROL: apply_temperature_limiting(); break;
  }
  
  // speed limit
  apply_speed_limit();
  
  // check if to enable the motor
  if ((!ui8_motor_enabled) &&
      (ui16_motor_get_motor_speed_erps() == 0) && // only enable motor if stopped, else something bad can happen due to high currents/regen or similar
      (ui8_adc_battery_current_target) &&
      (!ui8_brakes_engaged))
  {
    ui8_motor_enabled = 1;
    ui8_g_duty_cycle = 0;
    motor_enable_pwm();
  }

  // check if to disable the motor
  if ((ui8_motor_enabled) &&
      (ui16_motor_get_motor_speed_erps() == 0) &&
      (!ui8_adc_battery_current_target) &&
      (!ui8_g_duty_cycle))
  {
    ui8_motor_enabled = 0;
    motor_disable_pwm();
  }
  
  // reset control parameters if... (safety)
  if (ui8_brakes_engaged || ui8_system_state != NO_ERROR || !ui8_motor_enabled)
  {
    ui16_controller_duty_cycle_ramp_up_inverse_step = PWM_DUTY_CYCLE_RAMP_UP_INVERSE_STEP_DEFAULT;
    ui16_controller_duty_cycle_ramp_down_inverse_step = PWM_DUTY_CYCLE_RAMP_DOWN_INVERSE_STEP_MIN;
    ui8_controller_adc_battery_current_target = 0;
[... 64 lines cut ...]
                                               (uint32_t) ui16_duty_cycle_ramp_up_inverse_step_default,
                                               (uint32_t) PWM_DUTY_CYCLE_RAMP_UP_INVERSE_STEP_MIN);
                                               
    ui16_duty_cycle_ramp_down_inverse_step = map((uint32_t) ui16_wheel_speed_x10,
                                                 (uint32_t) 40, // 40 -> 4 kph
                                                 (uint32_t) 200, // 200 -> 20 kph
                                                 (uint32_t) PWM_DUTY_CYCLE_RAMP_DOWN_INVERSE_STEP_DEFAULT,
                                                 (uint32_t) PWM_DUTY_CYCLE_RAMP_DOWN_INVERSE_STEP_MIN);
                                                 
    // set battery current target
    if (ui16_adc_motor_current_target_torque_assist > ADC_10_BIT_MOTOR_PHASE_CURRENT_MAX) { ui8_adc_motor_current_target = ADC_10_BIT_MOTOR_PHASE_CURRENT_MAX; }
    else { ui8_adc_motor_current_target = ui16_adc_motor_current_target_torque_assist; }
        
    // set max battery current relative to the motor max current, this way it is a smooth transition from motor current control to battery current control
    ui8_adc_battery_current_target = fmapf((uint32_t) ui8_adc_motor_current_target,
                                           (uint32_t) 0,
                                           (uint32_t) ADC_10_BIT_MOTOR_PHASE_CURRENT_MAX,
                                           (uint32_t) 0,
                                           (uint32_t) ui8_adc_battery_current_max);

    // set duty cycle target
    if (ui8_adc_motor_current_target && ui8_adc_battery_current_target) { ui8_duty_cycle_target = PWM_DUTY_CYCLE_MAX; }
    else { ui8_duty_cycle_target = 0; }
  }
}`

AckmaniacSoftware, 2019-09-04:
please have a look at this commit, did not compile it but i hope you understand the idea
https://github.com/AckmaniacSoftware/TSDZ2-Smart-EBike/commit/02f58d07f6af2e474ab32d26adc88f58ac4130a8

casainho, 2019-09-05:
Currently, this is not a priority for me. I will back to this later. Thank you.

Frans-Willem, 2019-09-05:
Trying to look through your code, and I have one request:
If you are asking people to review, please clean up your commits. Make sure that all the green/red stuff I see is relevant to the change you're making. Doing this will greatly increase your chances of people actually taking the time to look through your proposed changes.

Some examples of things that (in my opinion) should've been cleaned up:
- https://github.com/AckmaniacSoftware/TSDZ2-Smart-EBike/commit/02f58d07f6af2e474ab32d26adc88f58ac4130a8#diff-9084047c203de63789e91b8491b59df5R368
  You changed one number, meaning I'd like to see one tiny bit of green indicating what you changed.
  Instead, I see one line removed, two added, one commented out duplicate line, and I have to actively search for what you changed and what your intention was.
- https://github.com/AckmaniacSoftware/TSDZ2-Smart-EBike/commit/02f58d07f6af2e474ab32d26adc88f58ac4130a8#diff-9084047c203de63789e91b8491b59df5R384
  You probably tried to change this, but instead ended up keeping it as-is. I'd recommend while creating your commit to just revert these lines before committing.

Apart from this, I haven't had the time to look in-depth at your proposal, but will try to later.

casainho, 2019-11-06:
Let's see if this goes to future V0.21.0.