OpenSourceEBike/TSDZ2-Smart-EBike #116: compilation on linux

benno90, 2019-09-15:
I installed installed the sdcc compiler and [stm8flash](https://github.com/vdudouyt/stm8flash) in order to compile the TSDZ2-Smart-EBike firmware. I'm currently stuck with the following error message `make: stm8-size: Command not found`. 
The makefile refers to another utility that i am probably missing: stm8-objcopy.
Is there an easy way to install these utilties? I cam across the following [page](https://stm8-binutils-gdb.sourceforge.io/) but i could not compile it.

Frans-Willem, 2019-09-15:
Had the same problems, take a peek at my repository and the Travis build instructions, should contain everything you need:
https://github.com/Frans-Willem/TSDZ2-Clean-EBike/blob/master/.travis.yml
https://github.com/Frans-Willem/TSDZ2-Clean-EBike/tree/master/travis

Mind you though, these scripts were written to be run in a temporary (throw-away) build environment. If you intend to install this stuff on your own machine, it might be better to install it somewhere in your home directory instead of root.

benno90, 2019-09-16:
When i try to compile the stm-binutils, the compilation crashes: `ser-tcp.c:64:13: error: conflicting declaration ‘typedef int socklen_t’`

benno90, 2019-09-17:
Definitely not able to compile stm-binutils on ubuntu18.
Tried it with two compiler versions: gcc-7 and gcc-4.8

benno90, 2019-11-13:
I had to remove my bashrc temporarily in order to compile the stm-binutils.
After that it worked.

closed 2019-11-13 by benno90