OpenSourceEBike/TSDZ2-Smart-EBike #118: TMP36 vs LM35 temperature sensor

j0bro, 2019-09-21:
While going through my stash of components I found a TMP36 sensor which is package and pin compatible with the LM35 but apparently slightly less accurate and the analog Vout translates to a different temperature offset: https://www.kevinsidwar.com/iot/2017/7/24/tmp36-vs-lm35-temp-sensor
They both however increase their output at 10mV/C. Is this sensor usable (maybe with a different offset in code)?

casainho, 2019-10-11:
No one seems to know. Closing for now.

closed 2019-10-11 by casainho

pklapperich, 2022-07-19:
From a glance at the datasheet it looks like the TMP36 will produce a 500mV offset at 0C, but still does 10mV/C just like the LM35/TMP35.

As long as the micro is OK with 1.5v on the input we're using then we can still read to 100C

This would be a nice software upgrade right now due to the IC shortage. TI has been sold out of LM35 for quite a while now and there's no end in sight.