OpenSourceEBike/TSDZ2-Smart-EBike #122: motor code should include a bit (or packet count) that indicates it is receiving packets from the display...

geeksville, 2019-10-03:
From a discussion with @r0mko:

> yes, if the tx line from your display to the rx line of the motor was broken I think that might explain it. and if it does, it would be great to put in an issue for the motor code to include a status bit in one of the packets it sends to the display that says 'is receiving packets from the display' - so we can detect this fault and tell the user ;-)

Does anything like this already exist.  If not it would be a great easy addition to the motor side so we could tell the user if we think there is a broken wire from the display to the motor.

If someone adds this, please pass this bug back to me and I promise to add display code to check for this bit ;-)

casainho, 2019-10-03:
Good idea. Even we need more, we need to separate configuration packets from regular packets, because configuration data will increase a lot with the points for the torque sensor calibration.

Ornias1993, 2022-10-14:
Note:
It would be prefered to only do these calculations in something like a dev-mode...