OpenSourceEBike/TSDZ2-Smart-EBike #123: Implement again startup BOOST

casainho, 2019-10-11:
[quote=Nfer post_id=1502836 time=1570650190 user_id=67657]
has the startup boost funcionality been removed from 0.20? I don't see it in the 0.20 configuration manual on the wiki
[/quote]
Buba removed that feature from 0.20.0 but I plan to implement it again on 0.21.0. In fact, I created an issue so I will not forget this.

Ornias1993, 2021-01-05:
Considering the legal limits in my country (the Netherlands) wattage limits are considered to be "continuous power", It would even be acceptable to have the "startup BOOST" cutoff speed configurable, that way people might be able to utilise a little more power when going up a hill, without needing to resort to off-road-mode or going over the speed limit.

Ornias1993, 2021-01-05:
I've looked into the relationship between maximum speed during boost mode, `maximum continuous rated power` (in accordance with EU Regulation 168/2013) and boostmode.

The `maximum continuous rated power` should be calculated based on EN:60034-1
A summary of which has been provided in EN:15194-2017:

> The maximum continuous rated power shall be measured according to EN 60034-1 when the motor
> reaches its thermal equilibrium as specified by the manufacturer.
> NOTE Thermal equilibrium: temperatures of motor parts do not vary more than 2K per hour.
> In circumstance where the power is measured directly at the shaft of the electronic motor, the result of
> the measurement shall be divided by 1,10 to consider the measurement uncertainty and then divided
> by 1,05 to include for example the transmission losses, unless the real values of these losses are
> determined.

This might seem bad, because the TSDZ2 (especially the higher end models) have much higher actuall "maximum continuous rated power", if one just measures temperature equilibrium and the actual physical motor speed.

However:
A "boost" mode, is never continuous output, as it's either limited by either speed or time. So is exempt from this regulation.
Linking boostmode to 6 kmh, would mean that boostmode could be enabled during heavy up-hill parts where the speed drops below 6kmh. This would prevent people from "falling off" their bike because the they slow down too much.

Linking it to the speed of the bike (actual speed) also techinically nukes the testing protocol, because the protocols do not take speed-based-overrides into account anyway.

**So my suggestion would be: Comply with N:15194-2017 and link boost mode to kmh.**

Another section where we could (and maybe even should) use boostmode with a time limit, is a time-limited boost based on user applied power. This would enable us to use road-legal mode during mountainbiking, as it would correct for the insane variances in power (and this is also how companies like Bosch circumvent the eu restrictions for their emtb's)
For reference:
https://fascatcoaching.com/tips/mountain-bike-power/