OpenSourceEBike/TSDZ2-Smart-EBike #124: Flash Program Memory

benno90, 2019-10-26:
In your compile and flash script you use the following device Type: -Device=STM8S105x6
which has 32kB of flash memory.
On the micro controller we found the type 'STM8S105 S4T6C'. According to the manual
and several online resources we found that this type of the STM8 MCU should have only 16kB
of flash memory. The OpenSource Ebike firmware uses approximately 26kB whereas
the original software fits into 16kB. 
Can you confirm that the micro controller in the TSDZ2 motor really has 32 kB?

casainho, 2019-10-26:
You are correct however this is a known issue with some ST microcontrollers. Please read about the same thing on the famous STM32F103.

The original firmware has near the 16 kbytes flash but we decided to go to the max 32 kbytes. I tested to write more than 32 kbytes and fails, the real max is the 32.

closed 2019-10-26 by casainho

benno90, 2019-10-26:
Thanks for the quick response. I made the same observation but the fact that all data sheets specify 16k memory left me somewhat confused.
I have an additional question: you dont seem to overwrite the data memory and option byte.  So
I assume that we just keep the data of the original firmware in this section (if there is anything there in the first place)?

casainho, 2019-10-26:
I think I don´t understand your question. I can say that option bytes are written by the firmware itself to make sure that some PWM channels are enabled - you need to look at the sources.

benno90, 2019-10-26:
I see, the option bytes can be altered during runtime.
The stm8s105 manual distinguishes data memory area and program memory area.
Using the stvp tool i can read out some values from the data memory (1kb according 
to the stm8s105 manual).  This section is never overwritten, only the program memory area.
So i was curious if the values in the data memory even matter.

casainho, 2019-10-26:
I think that yes, we use the EEPROM.

benno90, 2019-10-27:
Thank you, I understand now that you write settings such as the wheel perimeter to the eeprom section
of the memory. Do you recommend to clear the eeprom and option byte before installing your software or does it not matter ?