OpenSourceEBike/TSDZ2-Smart-EBike #126: Lights start blinking after a while (unsafe!)

j0bro, 2019-11-07:
Since end of October, wintertime started again here so the sun goes down early and I have to bike home in the dark; this means I need the lights now.

However, on both v0.19 and v0.20, the lights start blinking with a duty cycle of about 3 seconds after about 5 minutes or so and I can't get them to stay on continuously anymore, very unsafe! 

I did install the motor temperature sensor, nothing out of the ordinary (around 40C degrees), so it can't be a warning. But even after switching lights off and on, or even power cycling the bike, the blinking does not stop.

I hoped the new lighting modes in v0.20 would fix it, but the issue remains. Any pointers? Thanks!

Ederborg, 2019-11-07:
I use my bike and the lights every day. For me it has been working perfectly. I have the LCD3 version 0.20 and i have a temp sensor installed. Could it be something else that causes your problem? I don't think the temp alarm flashes the lights? I have had the alarm go off lots of times :)

j0bro, 2019-11-07:
No the temp alarm is only visible on the LCD, where the bottom right value for the motor temperature starts flashing whenever it hits the set limit.
My setup is with a 36V Samsung battery. My wife has the exact same motor + battery + lights, but with the VLCD5 stock display and for her it is also working fine. 

The strange thing is that it works for a few minutes, I ride for about 2 kms and then the slow flashing starts, the light will be on for about 3s and then off for 3s, and it repeats. No clue..!

Ederborg, 2019-11-07:
Strange. Could it be that the light draws to much power? I have a front and rear light intended for use with a 6v dynamo, i think it draws 1-3w or something.

j0bro, 2019-11-07:
I just checked on the LCD, when the lights are enabled (indeed also 6V ones), they draw 0.2-0.3 amps and the motor wattage indicator jumps between 0 and 10W so I guess that is very coarse rounding going on since 0.3A at 6V means only 1.8W in total. Will ride with lights on in the morning tomorrow and check what I can deduct from actual values of current/temperature/SoC/etc.

j0bro, 2019-11-16:
I have been riding around with different settings; I had one run where when I limited the amps to I think 10 or 12 instead of the default 16, the effect was not there. I had the slow flash again yesterday and it started after about 1 km where my daughter hung onto my arm on her normal bike. Whenever the issue comes in, there is no way for me to get back to the normal continuous light mode. When I came home I checked the timing of the slow flashing, it is ON for 3s and OFF for 2s, repeating this pattern.

leon927, 2019-11-18:
If your lights are pulling 0.3 A from the battery that is a lot of power. 

Your battery has a nominal voltage of 36 V. That would mean you are using lights that pull at least 10 W when pulling 0.3 A from the battery. This equals 1.7 A at 6 V on the light circuit.

Please use lower power lights and I guarantee you it will work.

KodeZ, 2020-07-21:
I think the lights are specced to be <5W. This type of flashing is somewhat typical for an overheating voltage regulator. LED lights tends to fail in this way too. If you are using a relay, and the motor light output as a control signal, then it sounds like a LED issue. If you are using the 6V directly from the motor, it could be the 36->6V conversion that is overheating. By duty-cycling it is effectively reducing the power draw, thus also the generated heat.
Two suggestions: Try a different light with lower wattage, or switch to a relay controlled light.
The relay controlled light is super easy to hook up, and is certainly the best option if you need the lights regularly and could use good lighting.

Linasfirebird, 2021-11-22:
Hi Everyone,

I am having the same issue with blinking lights after some minutes of driving. Very annoying. My set up is dual led front and single led in the back. I have no issues with such set up on my two other bafang motor bikes. Did you manage to solve the problem?

j0bro, 2021-11-22:
I suspect my lights drew too much current, which caused the capacitor to enter a drain/charge cycle causing the blinking.
Solved it by wiring my lights directly to the battery (yes, they can handle high voltages) and putting a [IRLZ44N MOSFET](https://opencircuit.nl/product/IRLZ44N-MOSFET-55V-47A) inbetween as the switch. The motor controller now controls the FET's gate with 6V and the lights run without issue now 👍🏻

Linasfirebird, 2021-11-22:
Hi Jeroen,

Thank you for quick response. So you have the lights that handle 6-55v?  I
have regular 6v led lights and my system is runing on 52v battery, so
probably would need step down converter anyway. But you other part of
solution with mosfet- is a good advice, thanks.