OpenSourceEBike/TSDZ2-Smart-EBike #127: new feature: mixed mode (power and torque)

bart594, 2019-11-16:
Hi,
We have power and torque mode, so maybe the combination of both like in Brose motors?
It could work based on rpm where you set up cadence up to bike works in torque mode and above in power mode. 
This could even replace boost mode by setting up higher assist level for torque mode  and lower for power mode so you have more power from the start and while speeding up with low cadence.
It can of course be useful when you want to have more control on technical trails and more power while riding with higher cadence. Then you set up the other way around lower assist level in torque and higher in power mode  

Here is description how it works in levo bikes. I have never ridden levo bike so i don't know how it feels but it looks like we have already all modes apart from "shuttle mode"
[https://www.emtbforums.com/community/threads/understanding-support-vs-peak-power-in-mission-control-app-v2-0.4038/page-2#post-73299](url)

>  the Brose Drive S Mag motor has a new feature from Brose that they call 'Flex Power Mode', essentially power delivery is controlled according to pedal force (torque - as per our previous gen Levo bikes) and also now cadence. This is hardware related - it's unique to this new motor on the 2019 Levo FSR. As a result, you get a significantly faster response from a standing start and thus the new Levo is much quicker when pulling away or starting up a steep hill compared to most other bikes. Mission Control "sees" that a new Levo is connected and displays Shuttle mode, enabling you to slide between 0% support (fully torque controlled) and 100% (overall power controlled, which is a combination of torque and cadence). The slider position determines the blend of torque and cadence as you mention above.

Ederborg, 2019-11-16:
This looks nice! I hope it will be implemented! :)

leon927, 2019-11-18:
That is the super riding mode I proposed on the Endless Sphere thread! It really is the best riding mode! Highly customizable and perfect for all users. It removes the need for all patches and Boost mode.

A very good suggestion @bart594 ! Every other riding mode falls short and is lacking in some way but this riding mode would cover everything.