OpenSourceEBike/TSDZ2-Smart-EBike #130: I'm confused about the release versions

kaelsaes, 2020-01-07:
Hi guys,

I would like to flash my TDSZ2 motor and my new 850C Display with the best firmware.
Where can I find the actual releases?

Why is there a TDSZ2_0.50 and a TDSZ2_0.19 release available?

Sorry, but it's hard for my to find the right versions.

Thanks in advance.

daenny, 2020-01-07:
Agreed, I also find this is very confusing. @leon927 did quite some improvements on TSDZ2_v0.20beta. Are they now (re-)implemented in the TSDZ2 0.51 version? How does the 0.51, 0.19 and 0.20beta version relate?
@casainho could you please clarify this?

casainho, 2020-01-07:
I advanced  to 0.50.0 so there are room if 0.20.0 want to increase.

0.50.0 has new features that none versions had but also misses features from 0.20.0. For various reasons I will not implement them and I see some of them as experimental. But I want to implement over time the features I think are relevant -- note that I am implementing for myself, what brings value for myself and not for others, do not expect I will implement, test and maintain features I do not use myself.

kaelsaes, 2020-01-07:
Okay, thanks for the information.

But what are the actual, stable versions for the TDSZ2 and the 850C?

casainho, 2020-01-07:
> Okay, thanks for the information.
> 
> But what are the actual, stable versions for the TDSZ2 and the 850C?

Follow the wiki page that says how to install 850C firmware, then on the releases link on github, it is the latest -- read the notes.

kaelsaes, 2020-01-21:
What's the difference between the files with and without bootloader?

kevin-jupiter, 2020-02-02:
Which branch is the 0.5x.x development work occurring in? I don't see any tags corresponding to versions 0.50 +.

kaelsaes, 2020-02-02:
@casainho 

What's the difference between the files with and without bootloader?

vshitikov, 2020-03-22:
Hi @casainho. Great job with the firmware.  I have a question, what commit that corresponds to the 0.54 firmware. I can see a tag for the 0.55 but there isn't such a tag for the 0.54. Can you clarify please. thank you.

casainho, 2020-03-22:
Well, I should made a release but I didn't. Look at the commit date and release date, you should get it.

wwwebrat, 2020-06-03:
maybe a little bit off topic as my problem is not with 850C (delete my comment if you think it disturbs).
I'm also confused about versioning.

Just to describe my problem:
I'm a user of the (apparently now deprecated) LCD3. Installed firmware some time ago and stopped by to check if there is a new firmware available. Instantly I was very confused where to look. 
1.0 is not suited for my hardware as I think to know now? Also the 0.57.x versions are not suitable for me? 
It seems that I have to use v0.20.0-beta.1?
Than I had a problem to find a the installation and configuration instructions as the wiki only shows instructions for the other versions and displays at first sight. (although the changelog contains a link to the 0.20.0)