OpenSourceEBike/TSDZ2-Smart-EBike #133: Expiremental motor type actually doesn't change the maximum RPM

r0mko, 2020-03-31:
Maximum motor RPM is always the same regardless of motor type settings in Motor submenu. The motor refuses to get past 96 RPM cadence despite it still delivers solid torque.

Demonstration: https://youtu.be/_K5zE5OjHYo

TheOriginalMrWolf, 2020-04-18:
I have the same experience.

I bought a 36v motor and 850C display from eco-ebike and coupled that with my 48v battery.  Display and motor were pre-programmed by EB.  Display shows v0.6.10.

Out-of-box, once I set the firmware to exp 36v, bike would happily spin up to about 110 rpm cadence.

Once I did the torque sensor calibration, reduced PAS levels to 9, set PAS levels, enabled and set BOOST, set battery details, enabled motor temp sensor (& set limits to 70C/80C), etc etc then suddenly it doesn't want to go above about 96 RPM.  I can get it above that if I pedal hard but the motor contribution is very small (drops to something like 60W).  I notice that the Cadence and Motor RPM indicators start flashing red.  Cadence about 96, motor RPM about 500.

Everything else is working perfectly, there's no excessive current draw, battery levels (volts, remaining capacity, etc) are fine, motor temp is good (eg 35C), and the bike rides great.  Just doesn't want to spin up above 96, and no real difference between motor: 36v and exp 36v.

Anyone got any hints?

TheOriginalMrWolf, 2020-04-18:
Ahhhhh... is this related to Pull Request #134 (https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/commit/8fbf63c3cb15bc5d72d25033927aea6151673979) ??

TheOriginalMrWolf, 2020-04-18:
Yes, indeed it is.

@r0mko - I've read through relevant code for display and motor, and can confirm that motor code prior to that pull request will only recognise standard 36v or 48v settings regardless of what you set in the display (both display:"36v" and display:"exp 36v" are interpreted by motor as "36v", and display:"48v" and display:"exp 48v" are interpreted by motor as "48v").

Simplest way to fix your issue is to update display and motor firmwares.

Unfortunately I can't prove by testing as I'm still waiting for the bootloader programmer for my display to be delivered - but my code readthrough confirms this issue & resolution.

r0mko, 2020-04-18:
Fixed in https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/pull/134

closed 2020-04-18 by r0mko