OpenSourceEBike/TSDZ2-Smart-EBike #139: (Still) no pedaling support when light's are on.

andrevil, 2020-08-19:
Hello together, 

I've recognized, that the pedal assist stops when the lights are on. It only stops when a real current is drawn. The consumption of my lamp is ~0.5A at a stable voltage of 6V. With light settings on but no light connected everything runs fine. 

I've seen there was a ticket in the past concerning this, but for me it is still not working. 

I've tried different "ADC lights offset / ADC lights " in the settings. But none of the values (0 to 4) helped. 

When I've looked in the motor.c I couldn't find anything concerning phase current calculation and light. Is this finding still fixed in this newest version?

BTW: This motor controlling firmware is very awesome and I have built up the second bike with it now. Thank you for this!

casainho, 2020-09-07:
@andrevil Can you please tell how is this going? is it working ok?

andrevil, 2020-09-15:
It seems to work without Problems. tested ca 60km

It's not a very nice solution. What about a battery current offset calibration after turning the light on?

QwertV2, 2020-09-17:
I got this exactly issue with the latest firmware and 860LCD. Previously I the emmebrusa mod, and lights always worked.
In that code I notice some difference on variable size and rotating bits.

Emmebrusa:

`  // calculate motor phase current ADC value
  if (ui8_g_duty_cycle > 0)
  {
    ui8_adc_motor_phase_current = (ui16_adc_battery_current << 6) / ui8_g_duty_cycle;
  }
  else
  {
    ui8_adc_motor_phase_current = 0;
  }`

Official:
 `// calculate motor current ADC value
  if (ui8_g_duty_cycle > 0)
  {
    ui16_g_adc_motor_current = ((ui16_g_adc_battery_current << 8) / ((uint16_t) ui8_g_duty_cycle));
  }
  else
  {
    ui16_g_adc_motor_current = 0;
  }`

I will try to use the suggested patch of andrevil, but I'm intrested to know what are the overall impacts.

ps: what about the solution provided in this opened pull request? is better?
https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/pull/128

Thank you guys for your AMAZING work! ;)

cookejames, 2020-09-22:
I'm also having this problem. I haven't tried your fix @andrevil but I'll give it a go if it isn't being merged into master as with autumn coming I need my lights again.

eraue, 2020-09-22:
Hello,
First of all. THANK YOU for this great soft.
Built an e-bike with a tsdz2 and 860c display that I use to commute 40 minutes to work every day.

I also have this issue and would very much appreciate a fix.

Thank you
[image]

NicobobinusX, 2020-09-27:
I've applied a fix that _seems_ to be working.  #141.  

I felt this performed better than the simple fix proposed earlier in this thread by Andrevil (https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/issues/139#issuecomment-678766584).  

Note that this is based on the current v1.0 code, where the other outstanding pull request from January seems to be based on an older version.  

It took me a while to get set up to where I could compile this and understand some might struggle even more than I did, so I am attaching a file here that includes the HEX I flashed to my motor should you want to do the same.  The only edits from the base v1.0 code are what I have in the pull request.  Included the motor.c file I edited in case that helps anyone.  Use at your own risk!
[lightsfix#141.zip](https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/files/5289090/lightsfix.141.zip)

NicobobinusX, 2020-09-28:
I tried different fixes from various threads for this issue and have found the one I've entered in pull request #142 to feel the most responsive.  It's really just a copy of #128 using the variable names from the V1.0 release.  It took me a while for me to get an environment set up where I could actually compile this successfully, so in case it helps anyone I'm attaching a ZIP file containing the compiled HEX file with this change, along with a copy of the edited motor.c source file shown in #142 to this post.  Use at your own risk, etc.

ZIP file:  [TSDZ2-v1.0_lightfix.zip](https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/files/5289700/TSDZ2-v1.0_lightfix.zip)

eraue, 2020-09-28:
> I tried different fixes from various threads for this issue and have found the one I've entered in pull request #142 to feel the most responsive. It's really just a copy of #128 using the variable names from the V1.0 release. It took me a while for me to get an environment set up where I could actually compile this successfully, so in case it helps anyone I'm attaching a ZIP file containing the compiled HEX file with this change, along with a copy of the edited motor.c source file shown in #142 to this post. While this is the exact same HEX file I flashed to my motor using STlink, use at your own risk, etc.
> 
> ZIP file: [TSDZ2-v1.0_lightfix.zip](https://github.com/OpenSource-EBike-firmware/TSDZ2-Smart-EBike/files/5289700/TSDZ2-v1.0_lightfix.zip)

Great, Thank you.
Will test it and keep you updated.

eraue, 2020-10-04:
Works perfectly.

Thank you again

eraue, 2020-10-07:
Hello again,

Tested more detailed today and looks like the assist level is 100% independent of the configuration and levels I use.
Also realized the motor current values displayed can't be accurate.  Very low compared to before.

BUT the light is working ;o)