OpenSourceEBike/TSDZ2-Smart-EBike #145: Wheel speed sanity check

iamblacklite, 2020-10-23:
Firmware is amazing 👍 Thankyou!

Only have one problem.  Occasionally I’m getting false readings from the speed sensor, with the display showing a speed in the 100’s.  The overwrites the assistance level on the display, and of course cuts out any motor assistance.  When it comes good the 3 digit incorrect speed is not fully cleared and the first digit continues to stay on the screen.  Would it be possible for the speed sensor to have a sanity check eg. If the value derived is over say 70-80km/h throw it away and don’t update it?

stefesciberras, 2021-05-14:
I second this. For some reason, my speed is detected as being >100, sometimes >w200. 

This only happens when i am pedalling, the speed sensor works great when coasting. 

Of course, when this happens, power is off, and it happens at the worst time: up a hill, coming out from an intersection etc

Ornias1993, 2022-10-14:
> Firmware is amazing 👍 Thankyou!
>
> Only have one problem. Occasionally I’m getting false readings from the speed sensor, with the display showing a speed in the 100’s. The overwrites the assistance [...]

This would take up already limited calculations, I think you might want to double check all connectors, as this sounds a bit like flickering a short in the speed sensor.