OpenSourceEBike/TSDZ2-Smart-EBike #146: EN:15194 Compliance

Ornias1993, 2021-01-05:
It would be nice to vett the performance of this firmware for EN:15194 compliance, or at least enable users to do so by giving clear instructions.
Mostly because the definitions for nominal rater power output are not put down in (european)-law, but put into the 100+ euro's costing EN:15194 standard.

If no one has access, i've a copy that might find it's way into your inbox by forwarding mistake somewhere these days.

casainho, 2021-01-05:
I am very busy with the wireless project.

This is DIY project and so I think there are no advantages to implement such standard. Unless is must be implemented, I don't see it to be a priority in the developers lists.

But thanks for sharing this information. The more you can share, the more later someone can learn on this and maybe implement it.

casainho, 2021-01-05:
Maybe this could be written to the wiki, as advice for the ones that want to configure the firmware in a way it follows most possible the standard.

Ornias1993, 2021-01-05:
I've read through EN:15194:2017 and I can not find any reason the current version doesn't comply with EN:15194:2017 already.
As listed above, there are only a few area's where one might not qualify for EN:15194:2017:
- Voltage mods
- motors _rated_ +250w.
- overspeeding

> Maybe this could be written to the wiki, as advice for the ones that want to configure the firmware in a way it follows most possible the standard.

Thats a great idea! 👍

tasakos10, 2021-03-25:
> It would be nice to vett the performance of this firmware for EN:15194 compliance, or at least enable users to do so by giving clear instructions.
> Mostly because the definitions for nominal rater [...]

Hello. Is it possible to forward me the EN15194/17 certificate? I have a TSDZ2 motor which has the 25km/h limitation as you already know, but the seller (pswpower) had only the certificate about electromagnetic radiation.
Thank you in advance

tasakos10, 2021-03-27:
> I've read through EN:15194:2017 and I can not find any reason the current version doesn't comply with EN:15194:2017 already.
> As listed above, there are only a few area's where one might not qualify for EN:15194:2017:
> 
> * Voltage mods
> * motors _rated_ +250w.
> * overspeeding
> 
> (which are mostly related to hardware choices or doing things that are **obviously** not a good idea anyway.)
> 
> > Maybe this could be written to the wiki, as advice for the ones that want to configure the firmware in a way it follows most possible the standard.
> 
> Thats a great idea! 👍

Hello. I have emailed you about that.
Thank you

Ornias1993, 2021-03-27:
Im going to close this issue, as it already documents what needs to be documented.

closed 2021-03-27 by Ornias1993