OpenSourceEBike/TSDZ2-Smart-EBike #150: tsdz2 with lcd3  problem

Karq81, 2021-03-22:
I just update my vlcd5 to lcd3 dispaly. But up and down buttom  not respond at all.  So i cant do any option change.  Had any one  same problem??? I try all firmware 0.19 and 0.20  same problem. I can only one the screen show everything  but no respond up/down ... :(((

papolak, 2021-07-04:
Hey, i had same issue, you probably have buttons connected to another legs of microcontroller, you just need to change in pins.h

Karq81, 2021-07-04:
Can you just explain a bit more with pins exactly meet to be changed or send a phot if you have one.
Thx for respond

OnVax, 2021-07-04:
[image]

OnVax, 2021-07-04:
I still have another issue with this LCD but buttons works

Karq81, 2021-07-04:
Omg this is to muche for me to change I will stay with original LCD with upgrade...

OnVax, 2021-07-04:
Contact me at email i will send you compiled files: onvax123@gmail.com

Karq81, 2021-07-04:
I did.