OpenSourceEBike/TSDZ2-Smart-EBike #151: Current controller counter issue in motor.c

4var1, 2021-04-12:
ui16_motor_speed_controller_counter never gets reset after firing so will fire lots more than expected... means it fires most of the time - only not firing for 2000 pwm cycles just after ui16_motor_speed_controller_counter wraps back to 0...

Not tested - just noticed as part of porting mspider65's changes across

    else if ((ui16_motor_speed_controller_counter > 2000) && // test about every 100ms
        (ui16_motor_speed_erps > ui16_max_motor_speed_erps))

The equivalent current counter gets reset ok - so would work as expected... 

    if (ui8_current_controller_counter > 14)
      ui8_current_controller_counter = 0;

casainho, 2021-04-12:
Good catch!

4var1, 2021-04-12:
no probs - any idea why I can't open issues for the Android App too?

casainho, 2021-04-13:
> no probs - any idea why I can't open issues for the Android App too?
Was disabled. You can now.

4var1, 2021-04-13:
OK great thanks