OpenSourceEBike/TSDZ2-Smart-EBike #160: TSDZ2-ESP-32 compatibility?

washichi, 2022-07-31:
I'm currently using this firmware for my TSDZ2, and would like to use the android app to change the configuration & monitor.
Does the emmebrusa firmware support the ESP32 bluetooth/wifi interface? Or are those projects existing next to each other?

I don't want to make the investment, to then find out it is not compatibly, and for sure someone here knows if it is compatible.

I'm talking about the TSDZ2-ESP32 project:
https://github.com/TSDZ2-ESP32/TSDZ2-ESP32-Wiki/

Ornias1993, 2022-10-14:
@mspider65 uses a seperate firmware, based on emmebrusa:
https://github.com/TSDZ2-ESP32/TSDZ2-ESP32-Wiki/

I think best choice is to ask him instead. As the maintainers of this repository have nothing to do with the TSDZ2-ESP32 project and it's compatibility.
But safe to assume it's not, as he is specifically building it's own firmware for it.

@casainho Mind to give me repo-wide write access (instead of just m500-m600 write access) so I can clean up the shittone of chores a bit?

casainho, 2022-10-14:
.You should got an invitation-

closed 2022-10-14 by casainho

Ornias1993, 2022-10-14:
Done thx! :)