OpenSourceEBike/TSDZ2-Smart-EBike #162: Battery Display

JasperCat123, 2022-10-20:
Hi,

I've got the 1.1 version installed and am using a Bafang colour display
The battery is a 48V 17.5
When I first set it all up it worked/displayed fine
When I install the fully charged battery the display shows 39%. I ride for a while, all good. The battery display goes down as it should.
When I stop, I turn the system off. Restart and it's back at 39% and the same thing happens again.
Eventually, it won't start up at 39% and at that point, I think I'm getting the true range
I'm still getting 65-70km range no problem.
Any ideas about what I need to do to fix it?

Kollins1982, 2022-12-30:
Hi Jasper,

I've experienced the same issue. A work around is to programm in the menu on the SOC site another value for the used Wh. If you do so the persisting value is overwritten and should be working after the next reset fine again. BR Kollins

JasperCat123, 2023-01-26:
Hi, thanks for that - I figured it was something pretty sipe as the battery and charger seem to be fine.
Any particular value - or just any value?

Kollins1982, 2023-01-26:
For me it worked to use any value. I think to overwrite the old value makes the difference.